SymbiFlow/vtr-verilog-to-routing #8: Placer cannot handle slice_l and slice_r carry chains in the same CLB

kgugala, 2019-02-10:
When two chains are packed into slice_r and slice_l of a single CLB placer fails with:

```
Error 1: 
Type: Placement
File: vtr-verilog-to-routing/vpr/src/place/place_macro.cpp
Line: 405
Message: Block #1 '$auto$alumacc.cc:474:replace_alu$14.slice[4].carry' appears in 2 placement macros (should appear in at most one). Related Macros:
  Macro #: 2
  Macro #: 3
```

kgugala, 2019-02-10:
This can be reproduced with https://github.com/SymbiFlow/symbiflow-arch-defs/pull/369

litghost, 2019-02-11:
This likely means that the placement macro matching is broken?  Likely not specific enough?

kgugala, 2019-02-19:
some digging gave the following:
Packer goes through all the molecules and tries to add them to a cluster. Here https://github.com/SymbiFlow/vtr-verilog-to-routing/blob/master%2Bwip/vpr/src/pack/cluster.cpp#L2240 it calls a function which is suppose to find a feasible molecule with the highest gain.

The `get_highest_gain_molecule` checks the feasiblity of each atom in a molecule using `exists_free_primitive_for_atom_block` (https://github.com/SymbiFlow/vtr-verilog-to-routing/blob/master%2Bwip/vpr/src/pack/cluster_placement.cpp#L709) which calls `primitive_type_feasible` (https://github.com/SymbiFlow/vtr-verilog-to-routing/blob/master%2Bwip/vpr/src/util/vpr_utils.cpp#L785) . This function checks only if the atom is the same model as currently considered and then if the ports and pins corespond between the cluster and atom.

In our case (specialized carry chains) this ends up in a situation where CARRY or CARRY0 from any chain is selected as a candidate for a cluster.

litghost, 2019-02-19:
> In our case (specialized carry chains) this ends up in a situation where CARRY or CARRY0 from any chain is selected as a candidate for a cluster.

If the pack pattern is working, the root of the chain in each tile should be a CARRY0.  Is that not happening?

kgugala, 2019-02-19:
> If the pack pattern is working, the root of the chain in each tile should be a CARRY0. Is that not happening?

It is happening - this is correct. 

The problem is that clusterer checks only if it is CARRY0 and packs unfeasible molecule into a cluster (e.g CC from BLK_TI-CLBLL_R.BLK_IG-SLICEL0.CARRYCHAIN to BLK_TI-CLBLL_R.BLK_IG-SLICEL1.CARRYCHAIN). This should be detected and such a cluster should get "unclustered", but it seems that the data gets corrupted and some molecules are not "unclustered". This seems to be another bug.

Those  accidentaly clustered molecules are not considered again, thus they in the end conflict with other properly clustered molecules.

Does that make sense?

litghost, 2019-02-19:
> The problem is that clusterer checks only if it is CARRY0 and packs unfeasible molecule into a cluster (e.g CC from BLK_TI-CLBLL_R.BLK_IG-SLICEL0.CARRYCHAIN to [...]

I thought #9 was supposed to have handled this (e.g. consuming entire chain).

kgugala, 2019-02-19:
#9 consumes whole carry chain and packs it into molecules. Those molecules are then passed to clusterer which confuses them.

kgugala, 2019-02-19:
one more thing: the error happens only when carry chain spans over more than one molecule

mithro, 2019-02-20:
Carry chains should never span more than one molecule?

kgugala, 2019-02-20:
Why? In our case carry molecule can have up to 4 atoms: 1 CARRY0 and 3 CARRYs. If a carry chain is longer than 4 (e.g. 5 bit counter) the chain will span over two molecules.

tmichalak, 2019-02-21:
I experimented with the chain_packing test and ran it with a design that contained three 8-bit counters.
The result of the clustering is as follows:
<pre>
block 0: $auto$alumacc.cc:474:replace_alu$17.slice[0].carry, type: BLK_TI-CLBLL_L
block 1: $auto$alumacc.cc:474:replace_alu$14.slice[0].carry, type: BLK_TI-CLBLL_R
block 2: $auto$alumacc.cc:474:replace_alu$14.slice[4].carry, type: BLK_TI-CLBLL_R
block 3: $auto$alumacc.cc:474:replace_alu$17.slice[4].carry, type: BLK_TI-CLBLL_L
block 4: $abc$198$n4, type: BLK_TI-CLBLM_R
block 5: $auto$alumacc.cc:474:replace_alu$11.slice[0].carry, type: BLK_TI-CLBLL_R
block 6: out:LED3
block 7: out:LED2
block 8: out:LED1
block 9: clk
</pre>
So what is interesting to notice is the lack of `$auto$alumacc.cc:474:replace_alu$11.slice[4].carry`
When digging a bit deeper into where it went I discovered that this chain molecule has been packed into cluster 2 along with `$auto$alumacc.cc:474:replace_alu$14.slice[4].carry`
I guess the question here is whether it should be packed in the same block? 

When I implemented a small hack which prevented  `$auto$alumacc.cc:474:replace_alu$11.slice[4].carry` from being packed into the same cluster the error went away and the clustering looked as follows:
<pre>
block 0: $auto$alumacc.cc:474:replace_alu$17.slice[0].carry, type: BLK_TI-CLBLL_L
block 1: $auto$alumacc.cc:474:replace_alu$14.slice[0].carry, type: BLK_TI-CLBLL_R
block 2: $auto$alumacc.cc:474:replace_alu$14.slice[4].carry, type: BLK_TI-CLBLL_R
block 3: $auto$alumacc.cc:474:replace_alu$17.slice[4].carry, type: BLK_TI-CLBLL_L
block 4: $auto$alumacc.cc:474:replace_alu$11.slice[4].carry, type: BLK_TI-CLBLL_R
block 5: $auto$alumacc.cc:474:replace_alu$11.slice[0].carry, type: BLK_TI-CLBLL_R
block 6: out:LED3
block 7: out:LED2
block 8: out:LED1
block 9: clk
</pre>

But this hack was just a way of checking whether the problem is before this stage or after.

What I did next was checking why the `$auto$alumacc.cc:474:replace_alu$11.slice[4].carry` was packed in block 2
gdb led me to `expand_pack_molecule_pin_edge` function.
Below is the complete stacktrace:
<pre>
#0  expand_pack_molecule_pin_edge (pattern_id=0, cur_pin=0x7fffffff9450, forward=77) at /home/tmichalak/Development/X-Ray/vtr-verilog-to-routing/vpr/src/pack/cluster_placement.cpp:604
#1  0x0000000000c29dc4 in expand_forced_pack_molecule_placement (molecule=0x1aa5f20, pack_pattern_block=0x1ab8330, primitives_list=0x1ae2380, cost=0x7fffffff9500) at /home/tmichalak/Development/X-Ray/vtr-verilog-to-routing/vpr/src/pack/cluster_placement.cpp:546
#2  0x0000000000c29b01 in try_place_molecule (molecule=0x1aa5f20, root=0x171bf90, primitives_list=0x1ae2380) at /home/tmichalak/Development/X-Ray/vtr-verilog-to-routing/vpr/src/pack/cluster_placement.cpp:486
#3  0x0000000000c28c75 in get_next_primitive_list (cluster_placement_stats=0x1acca70, molecule=0x1aa5f20, primitives_list=0x1ae2380) at /home/tmichalak/Development/X-Ray/vtr-verilog-to-routing/vpr/src/pack/cluster_placement.cpp:166
#4  0x0000000000c011f6 in try_pack_molecule (cluster_placement_stats_ptr=0x1acca70, atom_molecules=std::multimap with 57 elements = {...}, molecule=0x1aa5f20, primitives_list=0x1ae2380, pb=0x1d3cd60, max_models=26, max_cluster_size=176, clb_index=..., detailed_routing_stage=0, router_data=0x1b32080, debug_clustering=true, enable_pin_feasibility_filter=true, max_external_pin_util=...) at /home/tmichalak/Development/X-Ray/vtr-verilog-to-routing/vpr/src/pack/cluster.cpp:1307
#5  0x0000000000bfed3b in do_clustering (arch=0x7fffffffaa20, molecule_head=0x1acbf50, num_models=26, global_clocks=true, is_clock=std::unordered_set with 1 elements = {...}, atom_molecules=std::multimap with 57 elements = {...}, expected_lowest_cost_pb_gnode=std::unordered_map with 57 elements = {...}, hill_climbing_flag=false, out_fname=0x7fffffffa778 "top.net", timing_driven=true, cluster_seed_type=VPACK_TIMING, alpha=0.75, beta=0.899999976, inter_cluster_net_delay=5.42160001e+20, target_device_utilization=1, allow_unrelated_clustering=false, connection_driven=true, packer_algorithm=PACK_GREEDY, lb_type_rr_graphs=0x1939318, device_layout_name="xc7a50t-test", debug_clustering=true, enable_pin_feasibility_filter=true, ext_pin_util_targets=...) at /home/tmichalak/Development/X-Ray/vtr-verilog-to-routing/vpr/src/pack/cluster.cpp:634
#6  0x00000000009e2677 in try_pack (packer_opts=0x7fffffffa728, arch=0x7fffffffaa20, user_models=0x11689a0, library_models=0x1160630, interc_delay=5.42160001e+20, lb_type_rr_graphs=0x1939318) at /home/tmichalak/Development/X-Ray/vtr-verilog-to-routing/vpr/src/pack/pack.cpp:194
#7  0x00000000009709ad in vpr_pack (vpr_setup=..., arch=...) at /home/tmichalak/Development/X-Ray/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp:510
#8  0x000000000097060f in vpr_pack_flow (vpr_setup=..., arch=...) at /home/tmichalak/Development/X-Ray/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp:429
#9  0x000000000096fb64 in vpr_flow (vpr_setup=..., arch=...) at /home/tmichalak/Development/X-Ray/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp:303
#10 0x000000000095cb3b in main (argc=34, argv=0x7fffffffc758) at /home/tmichalak/Development/X-Ray/vtr-verilog-to-routing/vpr/src/main.cpp:69
</pre>
Note that for the lines to match check out the the carry_chains branch of our fork: https://github.com/antmicro/vtr-verilog-to-routing/tree/carry_chains

[... cut ...]
where `pattern_id` is the pack pattern assigned to the currently packed molecule and the `cur_pin` is of type `t_pb_graph_pin`.
I looked up in the code where the `pack_pattern_indices` are actually being set, which led me to `discover_pattern_names_in_pb_graph_node` in `prepack.cpp` which is being called only in one place which is `alloc_and_load_pack_patterns`. According to the comments its purpose is finding all packing patterns in the architecture.
Hence, my question at this point is whether there isn't something wrong with the pack patterns specified in the architecture xml?

litghost, 2019-02-21:
> I experimented with the chain_packing test and ran it with a design that contained three 8-bit counters.
> The result of the clustering is as follows:

The data you pasted in your comment does not appear to match the output from "pre_packing_molecules_and_patterns.echo".  Can you paste instances of pre_packing_molecules_and_patterns.echo from your various experiments?  I'm having trouble matching you data with your descriptions of behavior.

tmichalak, 2019-02-21:
Actually the contents are the same for both experiments:
<pre>
# of pack patterns 20
pack pattern index 0 block count 7 name LUT5toLUT7 root BEL_MX-F6MUX
pack pattern index 1 block count 15 name LUT5toLUT8 root BEL_MX-F6MUX
pack pattern index 2 block count 3 name LUT5toLUT6 root lut 
pack pattern index 3 block count 1 name LUT5x2 root lut 
pack pattern index 4 block count 4 name BLK_TI-CLBLL_R.BLK_IG-SLICEL0.CARRYCHAIN root BEL_BB-CARRY0
pack pattern index 5 block count 4 name BLK_TI-CLBLL_R.BLK_IG-SLICEL1.CARRYCHAIN root BEL_BB-CARRY0
pack pattern index 6 block count 4 name BLK_TI-CLBLL_L.BLK_IG-SLICEL0.CARRYCHAIN root BEL_BB-CARRY0
pack pattern index 7 block count 4 name BLK_TI-CLBLL_L.BLK_IG-SLICEL1.CARRYCHAIN root BEL_BB-CARRY0
pack pattern index 8 block count 4 name BLK_TI-CLBLM_R.BLK_IG-SLICEM.CARRYCHAIN root BEL_BB-CARRY0
pack pattern index 9 block count 3 name DRAM_DP root BLK_MM-DPRAM64
pack pattern index 10 block count 5 name DRAM_QP root BLK_MM-DPRAM64
pack pattern index 11 block count 7 name DRAM256 root BLK_MM-DPRAM64
pack pattern index 12 block count 3 name DRAM_DP_32 root BLK_MM-DPRAM32
pack pattern index 13 block count 5 name DRAM_QP_32 root BLK_MM-DPRAM32
pack pattern index 14 block count 3 name DRAM128 root BLK_MM-DPRAM128
pack pattern index 15 block count 3 name DRAM128_D root BLK_MM-DPRAM128
pack pattern index 16 block count 3 name DRAM128_DP root BEL_MX-F7AMUX
pack pattern index 17 block count 4 name BLK_TI-CLBLM_R.BLK_IG-SLICEL2.CARRYCHAIN root BEL_BB-CARRY0
pack pattern index 18 block count 4 name BLK_TI-CLBLM_L.BLK_IG-SLICEL2.CARRYCHAIN root BEL_BB-CARRY0
pack pattern index 19 block count 4 name BLK_TI-CLBLM_L.BLK_IG-SLICEM.CARRYCHAIN root BEL_BB-CARRY0

molecule type: atom
        pattern index 20: atom block $auto$simplemap.cc:420:simplemap_dff$99

molecule type: atom
        pattern index 20: atom block $auto$simplemap.cc:420:simplemap_dff$98

molecule type: atom
        pattern index 20: atom block $auto$simplemap.cc:420:simplemap_dff$97

molecule type: atom
        pattern index 20: atom block $auto$simplemap.cc:420:simplemap_dff$96

molecule type: atom
        pattern index 20: atom block $auto$simplemap.cc:420:simplemap_dff$95

molecule type: atom
        pattern index 20: atom block $auto$simplemap.cc:420:simplemap_dff$94

molecule type: atom
        pattern index 20: atom block $auto$simplemap.cc:420:simplemap_dff$93

molecule type: atom
        pattern index 20: atom block $auto$simplemap.cc:420:simplemap_dff$92

molecule type: atom
        pattern index 20: atom block $auto$simplemap.cc:420:simplemap_dff$91

molecule type: atom
        pattern index 20: atom block $auto$simplemap.cc:420:simplemap_dff$90

molecule type: atom 
        pattern index 20: atom block $auto$simplemap.cc:420:simplemap_dff$89 
 
molecule type: atom 
        pattern index 20: atom block $auto$simplemap.cc:420:simplemap_dff$88 
 
molecule type: atom 
        pattern index 20: atom block $auto$simplemap.cc:420:simplemap_dff$87 
 
molecule type: atom 
        pattern index 20: atom block $auto$simplemap.cc:420:simplemap_dff$86 
 
molecule type: atom 
        pattern index 20: atom block $auto$simplemap.cc:420:simplemap_dff$85 
 
molecule type: atom 
[... 107 lines cut ...]
Block_Id: 4, x_offset: 0, y_offset: -1, z_offset: 0
</pre>
and without it
<pre>
#Identified Placement macros
Num_Macros: 3
Macro_Id: 0, Num_Blocks: 2
------------------------------------------------------
Block_Id: 0, x_offset: 0, y_offset: 0, z_offset: 0
Block_Id: 3, x_offset: 0, y_offset: -1, z_offset: 0

Macro_Id: 1, Num_Blocks: 2
------------------------------------------------------
Block_Id: 1, x_offset: 0, y_offset: 0, z_offset: 0
Block_Id: 2, x_offset: 0, y_offset: -1, z_offset: 0

Macro_Id: 2, Num_Blocks: 2
------------------------------------------------------
Block_Id: 5, x_offset: 0, y_offset: 0, z_offset: 0
Block_Id: 2, x_offset: 0, y_offset: -1, z_offset: 0
</pre>
Note how the block_id: 2 is listed twice - for macro_id 1 and 2.
It seems that macro_id:2 is for packing the `$auto$alumacc.cc:474:replace_alu$11` carry chain.
Block_id:5 is actually `$auto$alumacc.cc:474:replace_alu$11.slice[0].carry`
With my hack Block_id:4 is `$auto$alumacc.cc:474:replace_alu$11.slice[4].carry` and without it it's `$abc$198$n4`. Moreover Macro_id:2 reports it is using Block_id:2. This would coincide with what I described in my previous post about the carry molecule `$auto$alumacc.cc:474:replace_alu$11.slice[4].carry` being packed into Block 2.

mithro, 2019-02-23:
Can you get the types for each of the pack patterns?

mithro, 2019-02-23:
Just restructuring these into how they should be looked at;

----

`$auto$alumacc.cc:474:replace_alu$17`
```
molecule type: BLK_TI-CLBLL_L.BLK_IG-SLICEL0.CARRYCHAIN 
        pattern index 0: atom block $auto$alumacc.cc:474:replace_alu$17.slice[0].carry root node 
        pattern index 1: atom block $auto$alumacc.cc:474:replace_alu$17.slice[1].carry 
        pattern index 2: atom block $auto$alumacc.cc:474:replace_alu$17.slice[2].carry 
        pattern index 3: atom block $auto$alumacc.cc:474:replace_alu$17.slice[3].carry 

molecule type: BLK_TI-CLBLL_L.BLK_IG-SLICEL0.CARRYCHAIN 
        pattern index 0: atom block $auto$alumacc.cc:474:replace_alu$17.slice[4].carry root node 
        pattern index 1: atom block $auto$alumacc.cc:474:replace_alu$17.slice[5].carry 
        pattern index 2: atom block $auto$alumacc.cc:474:replace_alu$17.slice[6].carry 
        pattern index 3: atom block $auto$alumacc.cc:474:replace_alu$17.slice[7].top_of_carry 
```

----

`$auto$alumacc.cc:474:replace_alu$14`
``` 
molecule type: BLK_TI-CLBLL_R.BLK_IG-SLICEL1.CARRYCHAIN 
        pattern index 0: atom block $auto$alumacc.cc:474:replace_alu$14.slice[0].carry root node 
        pattern index 1: atom block $auto$alumacc.cc:474:replace_alu$14.slice[1].carry 
        pattern index 2: atom block $auto$alumacc.cc:474:replace_alu$14.slice[2].carry 
        pattern index 3: atom block $auto$alumacc.cc:474:replace_alu$14.slice[3].carry 

molecule type: BLK_TI-CLBLL_R.BLK_IG-SLICEL1.CARRYCHAIN 
        pattern index 0: atom block $auto$alumacc.cc:474:replace_alu$14.slice[4].carry root node 
        pattern index 1: atom block $auto$alumacc.cc:474:replace_alu$14.slice[5].carry 
        pattern index 2: atom block $auto$alumacc.cc:474:replace_alu$14.slice[6].carry 
        pattern index 3: atom block $auto$alumacc.cc:474:replace_alu$14.slice[7].top_of_carry 
```

----

`$auto$alumacc.cc:474:replace_alu$11`
```
molecule type: BLK_TI-CLBLL_R.BLK_IG-SLICEL0.CARRYCHAIN 
        pattern index 0: atom block $auto$alumacc.cc:474:replace_alu$11.slice[0].carry root node 
        pattern index 1: atom block $auto$alumacc.cc:474:replace_alu$11.slice[1].carry 
        pattern index 2: atom block $auto$alumacc.cc:474:replace_alu$11.slice[2].carry
        pattern index 3: atom block $auto$alumacc.cc:474:replace_alu$11.slice[3].carry
 
molecule type: BLK_TI-CLBLL_R.BLK_IG-SLICEL0.CARRYCHAIN 
        pattern index 0: atom block $auto$alumacc.cc:474:replace_alu$11.slice[4].carry root node 
        pattern index 1: atom block $auto$alumacc.cc:474:replace_alu$11.slice[5].carry 
        pattern index 2: atom block $auto$alumacc.cc:474:replace_alu$11.slice[6].carry 
        pattern index 3: atom block $auto$alumacc.cc:474:replace_alu$11.slice[7].top_of_carry 
```

mithro, 2019-06-05:
This should be fixed by the #17 .

closed 2019-06-05 by mithro