SymbiFlow/vtr-verilog-to-routing #13: Add automatic C++ formatting and format upstream repository

litghost, 2019-03-04:
- [ ] Define clang-format rules for vtr C++
- [ ] Create target to automatically format the code
- [ ] Push new target upstream
- [ ] Push formatted code upstream

kgugala, 2019-03-12:
the formating now handles also libs/EXTERNAL folder. The code is here https://github.com/kgugala/vtr-verilog-to-routing/tree/code-format

here you can find make format results: https://github.com/kgugala/vtr-verilog-to-routing/tree/code-format-result-all

@mithro please check if the formatting is what you expected

<blockquote>[image]<div>[image] GitHub</div><div><strong><a href="https://github.com/kgugala/vtr-verilog-to-routing">kgugala/vtr-verilog-to-routing</a></strong></div><div>Verilog to Routing -- Open Source CAD Flow for FPGA Research - kgugala/vtr-verilog-to-routing</div></blockquote>
<blockquote>[image]<div>[image] GitHub</div><div><strong><a href="https://github.com/kgugala/vtr-verilog-to-routing">kgugala/vtr-verilog-to-routing</a></strong></div><div>Verilog to Routing -- Open Source CAD Flow for FPGA Research - kgugala/vtr-verilog-to-routing</div></blockquote>

mithro, 2019-06-05:
This has been done!

closed 2019-06-05 by mithro