SymbiFlow/vtr-verilog-to-routing #17: Implement equivalent tile types support

litghost, 2019-03-04:
@mithro to provide description

mithro, 2019-03-05:
The [design doc is currently here](https://docs.google.com/document/d/1LQ0H2C9IFnCVtSuaniCs1dDmBoMBWwBQLKAGuo_kSc0/edit#).

The general idea is to add a new "tile" tag. The clusterer continues to use the pb_type while the placer uses the tile tag to know which pb_types can be put at a given location. A bunch of the stuff which was only valid on the top level pb_type (such as pin sides) is moved to the tile type instead.

mkurc-ant, 2019-03-05:
This is related to the approach from https://github.com/SymbiFlow/symbiflow-arch-defs/issues/415. I think that the 415 should be done first before starting work with this one

mithro, 2019-03-05:
@mkurc-ant I would suggest they can be done in parallel if we can find people to work on it.

mkurc-ant, 2019-03-11:
I've given some thoughts to this issue and https://github.com/SymbiFlow/symbiflow-arch-defs/issues/415. I think those two are contradictory.

In 415 we want to modify the data that comes into the VPR so it contain already split CLBs. Therefore no modification to the VPR which introduces "tile_type" is necessary. I've verified this approach and it solves issue with carry chain packing. Not sure though what with placing SLICELs into SLICEMs but I guess that it can be solved by defining "L" mode for SLICEM.

On the other hand in this issue we want to have a grid with "tile_types" of CLBs in which every CLB has two slices. The VPR code needs to be modified so it can understand the new "tile_type" structure. So there is no need for changing input data to it. It needs to be rather re-formatted to use "tile_types", but no grid split.

What is the ultimate outcome we want to achieve ?

litghost, 2019-03-11:
> I've given some thoughts to this issue and [SymbiFlow/symbiflow-arch-defs#415](https://github.com/SymbiFlow/symbiflow-arch-defs/issues/415). I think those two are contradictory.
> 
> In 415 we want to modify the data that comes into the VPR so it contain already split CLBs. Therefore no modification to the VPR which introduces "tile_type" is necessary. I've verified this approach and it solves issue with carry chain packing. Not sure though what with placing SLICELs into SLICEMs but I guess that it can be solved by defining "L" mode for SLICEM.

This is what we want.
 
> On the other hand in this issue we want to have a grid with "tile_types" of CLBs in which every CLB has two slices. The VPR code needs to be modified so it can understand the new "tile_type" [...]

You are misunderstanding.  This issue is to have each grid have one slice, but in the case of the SLICEM, the supported pb_types in the grid location is either SLICEL or SLICEM.

> What is the ultimate outcome we want to achieve ?

The packer chooses the narrowest/least restrictive packing option, and the placer has the option to use either SLICEL or SLICEM depending on resource and timing.

mkurc-ant, 2019-03-12:
Thanks, I understand the case with packing SLICEL into SLICEM.

But regarding the split, the document https://docs.google.com/document/d/1LQ0H2C9IFnCVtSuaniCs1dDmBoMBWwBQLKAGuo_kSc0 states that there can be a tile with two slices. So if the grid is made of such tiles then the statement that "each grid tile has one SLICE" won't be true. At least not in the architecture XML file.

If we want to have a new "tile type" in the XML, which can have multiple sites for SLICEs, then we will have to modify the VPR to perform a kind of the grid split internally. Am I understanding this correctly? Is it the way to go ?

mithro, 2019-03-12:
@mkurc-ant The split and the equivalent placement types are independent of each other. 

The equivalent placement types should work with a single tile at a location or multiple tiles at a location.

litghost, 2019-04-01:
@acomodi's design docuement: https://docs.google.com/document/d/1lWLgZ5fCwKUgPh1MaOoAHwKBBr1ZAQLDzPkJtzTjLvQ/edit?ts=5ca2264c#heading=h.w3iqpzg5p7di

closed 2019-12-14 by litghost