SymbiFlow/vtr-verilog-to-routing #31: Round robin packing option introduces issues in the routing step

acomodi, 2019-03-22:
After the upstream merge introduced by https://github.com/SymbiFlow/vtr-verilog-to-routing/pull/30 `murax` and `picosoc` cannot be routed anymore.
The routing step incurs in an apparently infinite loop when trying to solve the `Overused RR nodes`, not being able to find a feasible solution.

Being Round Robin not necessary anymore after the tile split, this should not be a major issue, but it is worth investigating what causes this behavior.

Example of `infinite iterations`:

```
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
      (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter  
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
   1   35.8     0.0    0 1.8e+08    2631    8642    4659 ( 0.235%)  156658 ( 0.5%) 45223500000000.000 -4.173e+16 -45223500000000.000      0.000      0.000      N/A   
   2   71.2     0.5   65 2.9e+08    2067    7513    2070 ( 0.104%)  155784 ( 0.5%) 52946000000000.000 -4.441e+16 -52946000000000.000      0.000      0.000      N/A   
   3  101.1     0.6   47 3.6e+08    1601    6098    1444 ( 0.073%)  159057 ( 0.5%) 51292000000000.000 -4.424e+16 -51292000000000.000      0.000      0.000      N/A   
   4  112.6     0.8   43 3.8e+08    1456    5846    1026 ( 0.052%)  161168 ( 0.5%) 50738500000000.000 -4.430e+16 -50738500000000.000      0.000      0.000      N/A   
   5   93.6     1.1   55 3.2e+08    1229    5378     200 ( 0.010%)  163311 ( 0.5%) 56258000000000.000 -4.633e+16 -56258000000000.000      0.000      0.000      N/A   
   6   78.9     1.4   48 2.8e+08    1152    4982     125 ( 0.006%)  163935 ( 0.5%) 51293500000000.000 -4.549e+16 -51293500000000.000      0.000      0.000      N/A   
   7   71.2     1.9   18 2.5e+08    1109    4714      66 ( 0.003%)  163863 ( 0.5%) 50742500000000.000 -4.522e+16 -50742500000000.000      0.000      0.000      N/A   
   8   67.5     2.4   15 2.4e+08    1071    4552      12 ( 0.001%)  163696 ( 0.5%) 53497500000000.000 -4.488e+16 -53497500000000.000      0.000      0.000      N/A   
   9   63.9     3.1   10 2.2e+08    1055    4449       7 ( 0.000%)  164428 ( 0.5%) 57359000000000.000 -4.516e+16 -57359000000000.000      0.000      0.000      N/A   
  10   64.0     4.1   11 2.2e+08    1037    4361       7 ( 0.000%)  163512 ( 0.5%) 52946000000000.000 -4.454e+16 -52946000000000.000      0.000      0.000       11
  11   59.1     5.3    5 2.1e+08    1030    4302       5 ( 0.000%)  163441 ( 0.5%) 51843000000000.000 -4.420e+16 -51843000000000.000      0.000      0.000       12
  12   56.5     6.9    9 2.0e+08    1020    4265       5 ( 0.000%)  163571 ( 0.5%) 52394500000000.000 -4.447e+16 -52394500000000.000      0.000      0.000       13
  13   54.7     9.0    7 1.9e+08    1013    4231       4 ( 0.000%)  163386 ( 0.5%) 52394500000000.000 -4.411e+16 -52394500000000.000      0.000      0.000       15
  14   52.3    11.6    6 1.9e+08    1000    4190       3 ( 0.000%)  163258 ( 0.5%) 52946500000000.000 -4.416e+16 -52946500000000.000      0.000      0.000       16
...
 485   62.7 1.0e+25    0 2.0e+08     951    3880       1 ( 0.000%)  164538 ( 0.5%) 51292500000000.000 -4.354e+16 -51292500000000.000      0.000      0.000     1068
 486   61.1 1.0e+25    0 2.0e+08     951    3880       1 ( 0.000%)  164986 ( 0.5%) 49085500000000.000 -4.363e+16 -49085500000000.000      0.000      0.000     1027
 487   61.5 1.0e+25    0 2.0e+08     951    3880       2 ( 0.000%)  164921 ( 0.5%) 49638500000000.000 -4.328e+16 -49638500000000.000      0.000      0.000      949
 488   62.4 1.0e+25    0 2.0e+08     951    3880       1 ( 0.000%)  164774 ( 0.5%) 51292500000000.000 -4.349e+16 -51292500000000.000      0.000      0.000     1026
 489   61.9 1.0e+25    0 2.0e+08     951    3879       1 ( 0.000%)  164468 ( 0.5%) 51292500000000.000 -4.330e+16 -51292500000000.000      0.000      0.000      949
 490   60.0 1.0e+25    0 1.9e+08     950    3877       1 ( 0.000%)  164378 ( 0.5%) 50741500000000.000 -4.359e+16 -50741500000000.000      0.000      0.000      922
 491   62.1 1.0e+25    0 2.0e+08     950    3877       1 ( 0.000%)  164699 ( 0.5%) 50740500000000.000 -4.357e+16 -50740500000000.000      0.000      0.000      868
 492   60.5 1.0e+25    0 1.9e+08     950    3877       2 ( 0.000%)  164638 ( 0.5%) 50742000000000.000 -4.368e+16 -50742000000000.000      0.000      0.000      848
 493   60.6 1.0e+25    0 1.9e+08     950    3877       1 ( 0.000%)  164442 ( 0.5%) 49638500000000.000 -4.370e+16 -49638500000000.000      0.000      0.000      940
 494   65.9 1.0e+25    0 2.1e+08     950    3878       2 ( 0.000%)  164542 ( 0.5%) 49087000000000.000 -4.361e+16 -49087000000000.000      0.000      0.000      914
 495   63.5 1.0e+25    0 2.0e+08     950    3878       3 ( 0.000%)  165144 ( 0.5%) 50740500000000.000 -4.352e+16 -50740500000000.000      0.000      0.000     1034
 496   62.5 1.0e+25    0 2.0e+08     950    3879       1 ( 0.000%)  164879 ( 0.5%) 49638500000000.000 -4.346e+16 -49638500000000.000      0.000      0.000     1234
 497   62.0 1.0e+25    0 2.0e+08     950    3879       1 ( 0.000%)  164608 ( 0.5%) 51844000000000.000 -4.335e+16 -51844000000000.000      0.000      0.000     1275
 498   61.1 1.0e+25    0 2.0e+08     950    3878       1 ( 0.000%)  164731 ( 0.5%) 49638000000000.000 -4.322e+16 -49638000000000.000      0.000      0.000     1209
 499   65.2 1.0e+25    0 2.0e+08     951    3881       1 ( 0.000%)  164722 ( 0.5%) 50190000000000.000 -4.340e+16 -50190000000000.000      0.000      0.000     1250
 500   64.3 1.0e+25    0 2.0e+08     951    3881       1 ( 0.000%)  164523 ( 0.5%) 51843500000000.000 -4.360e+16 -51843500000000.000      0.000      0.000     1189
Routing failed.
Router Stats: total_nets_routed: 483294 total_connections_routed: 1977732 total_heap_pushes: 99403897290 total_heap_pops: 43086821171
# Routing took 30473.81 seconds (max_rss 5825.2 MiB, delta_rss +13.6 MiB)
Circuit is unroutable with a channel width factor of 500.
VPR failed to implement circuit

```

mithro, 2019-06-05:
@acomodi Are we just abandoning this?

acomodi, 2019-06-05:
@mithro Yes, I believe it could be closed. IMO the round robin patch as well. @litghost what do you think?

litghost, 2019-06-05:
> @mithro Yes, I believe it could be closed. IMO the round robin patch as well. @litghost what do you think?

Ya

acomodi, 2019-06-05:
Ok, when dealing with https://github.com/SymbiFlow/vtr-verilog-to-routing/issues/71 I'll discard RR as well (or at least place it in a branch just in case it could be useful in the future).

I'll close this issue then.

closed 2019-06-05 by acomodi