SymbiFlow/vtr-verilog-to-routing #32: hlc disabled

acomodi, 2019-03-22:
With https://github.com/SymbiFlow/vtr-verilog-to-routing/pull/30 `hlc` had to be disabled given that the `metadata` PR upstream  (https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/473) has made `hlc` incompatible.

I didn't further investigate the issue, but as far as I have understood, in #473 there was a switch from pointer to non-pointer when referencing to the `metadata` type, causing compile time errors when accessing to it.

elmsfu, 2019-03-22:
@acomodi genhlc is already not used in symbiflow-arch-defs I still need to get back to removing all the references to HLC SymbiFlow/symbiflow-arch-defs#493, but it's low priority and I think other tasks overlap with the work. If it's disable we can go ahead and remove the code.

acomodi, 2019-03-22:
@elmsfu Ok, by removing the code you mean deleting the whole `hlc` folder from VTR, right?

elmsfu, 2019-03-22:
@acomodi Yeah. If you prefer I can do it on a separate PR when I finish the other cleanup

acomodi, 2019-03-25:
@elmsfu I guess it would be better if I do it within PR https://github.com/SymbiFlow/vtr-verilog-to-routing/pull/30

mithro, 2019-06-05:
HLC has been disabled, we should remove it from the git history. See #71

closed 2019-06-05 by mithro