SymbiFlow/vtr-verilog-to-routing #43: Warnings while bulding

Prayas1997, 2019-04-20:


Prayas1997, 2019-04-20:
1.
/home/prayas/sandbox/vtr-debug/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp: In function ‘RouteStatus vpr_load_routing(t_vpr_setup&, const t_arch&, int, std::shared_ptr<SetupHoldTimingInfo>, vtr::vector<vtr::StrongId<cluster_net_id_tag>, float*>&)’:
/home/prayas/sandbox/vtr-debug/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp:773:68: warning: unused parameter ‘arch’ [-Wunused-parameter]
2.
/home/prayas/sandbox/vtr-debug/vtr-verilog-to-routing/vpr/src/pack/cluster_router.cpp: In function ‘bool check_edge_for_route_conflicts(std::unordered_map<const t_pb_graph_node*, const t_mode*>*, const t_pb_graph_pin*, const t_pb_graph_pin*)’:
/home/prayas/sandbox/vtr-debug/vtr-verilog-to-routing/vpr/src/pack/cluster_router.cpp:208:44: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

3.
/home/prayas/sandbox/vtr-debug/vtr-verilog-to-routing/vpr/src/base/read_netlist.cpp: In function ‘set_atom_pin_mapping’:
/home/prayas/sandbox/vtr-debug/vtr-verilog-to-routing/vpr/src/base/vpr_types.cpp:128:12: warning: potential null pointer dereference [-Wnull-dereference]
     return child_pbs == nullptr;
            ^
/home/prayas/sandbox/vtr-debug/vtr-verilog-to-routing/vpr/src/base/vpr_types.cpp:128:12: warning: potential null pointer dereference [-Wnull-dereference]

steps to reproduce: run make from vtr root directory

acomodi, 2019-04-20:
Please enclose snippets of code/logs with ```  to make them more readable.

For example it should look like this:
```
/home/.../vpr_api.cpp: In function... vtr::vector<vtr::StrongId<cluster_net_id_tag>, float*>&)’:
logs
logs
```

Prayas1997, 2019-04-20:
> Please enclose snippets of code/logs with ``` to make them more readable.
> 
> For example it should look like this:
> 
> ```
> /home/.../vpr_api.cpp: In function... vtr::vector<vtr::StrongId<cluster_net_id_tag>, float*>&)’:
> logs
> logs
> ```

Alright! Updated :)

mithro, 2019-06-05:
I believe this is no longer a problem?

closed 2019-06-05 by mithro