SymbiFlow/vtr-verilog-to-routing #71: Rework this repository

mithro, 2019-06-05:
- [ ] Make `master` equivalent to master on https://github.com/verilog-to-routing/vtr-verilog-to-routing. There should be no patches in this branch which are not upstream.
 - [ ] Make a branch in this repo for each feature which has **yet** to be merged. These branches will be continually rebased on top of upstream/master.
 - [ ] Make `master+wip` be an "octopus merge" of all the current pending branches + upstream `master`. This commit will be continually replaced.

The new update strategy should be;
 1) Update SymbiFlow/vtr-verilog-to-routing `master` branch to match upstream master.
 2) Rebase each unmerged branch on top of `master`
 3) Create a new `master+wip` head using the "octopus merge".
 4) Force push over the previous `master+wip` branch.

Example update process;
```
git remote add origin git+ssh://github.com/SymbiFlow/vtr-verilog-to-routing.git
git remote add upstream git+ssh://github.com/verilog-to-routing/vtr-verilog-to-routing.git
git fetch upstream
git checkout master
git reset --hard upstream/master
git push origin master

for BRANCH in IN_PROGRESS_BRANCH; do
  git branch -D $BRANCH
  git checkout origin/$BRANCH -b $BRANCH
  git rebase -i upstream/master
  git push origin $BRANCH --force
done

git checkout master+wip
git reset --hard upstream/master
git merge $IN_PROGRESS_BRANCH
git push origin master+wip --force
```

acomodi, 2019-06-05:
@mithro Great idea as `master+wip` is getting quite messy. 

arch-defs is still going to use master+wip, correct? At least until all the "critical" patches get merged upstream.

mithro, 2019-06-05:
Should be pretty easy to see the status of each feature as it will always be relative to `upstream/master`.

acomodi, 2019-06-06:
@mithro can you give me authorization to directly push on this repo, please? 
I get the following:
```
Counting objects: 25, done.
Delta compression using up to 20 threads.
Compressing objects: 100% (11/11), done.
Writing objects: 100% (25/25), 4.47 KiB | 4.47 MiB/s, done.
Total 25 (delta 19), reused 20 (delta 14)
remote: Resolving deltas: 100% (19/19), completed with 14 local objects.
remote: error: GH006: Protected branch update failed for refs/heads/master.
remote: error: You're not authorized to push to this branch. Visit https://help.github.com/articles/about-protected-branches/ for more information.
To https://github.com/SymbiFlow/vtr-verilog-to-routing.git
 ! [remote rejected]     master -> master (protected branch hook declined)
error: failed to push some refs to 'https://github.com/SymbiFlow/vtr-verilog-to-routing.git'

```

acomodi, 2019-06-06:
@mithro I believe that branches `generalized_pack_patterns`, `interposer` and `pre_hcl_separation` could be deleted, do you agree?

mithro, 2019-06-06:
@acomodi Check now?

acomodi, 2019-06-06:
@mithro @litghost I have created a `master+wip-backup` just in case something goes wrong. I have isolated all the patches in the various branches. Not all of them are going to be upstream, but right now they are well divided and easy to look at.

In my previous comment, I mentioned some branches which could be deleted IMO.

I am currently running locally the `all_xc7` and `all_ice40` SymbiFlow tests on the octopus merge master+wip branch. Once everything works I'll push force it overwriting the current one. Does it sound good to you?

mithro, 2019-06-06:
@acomodi Sounds good to me!

litghost, 2019-06-06:
> @mithro @litghost I have created a `master+wip-backup` just in case something goes wrong. I have isolated all the patches in the various branches. Not all of them are going to be upstream, but right now they are well divided and easy to look at.

Sounds good.  Once the octopus merge is ready, it would be good to have list of outstanding branches to triage.

mithro, 2019-06-06:
@acomodi I think we should create a github issue for each branch with info on;
 * If it needs to be merged upstream with a reference to the pull request when it happens.
 * If it needs to be abandoned / deleted and under what conditions that should happen.

acomodi, 2019-06-07:
@mithro Done, all branches are isolated and commented. master+wip now is cleaned and contains the octopus merge of all the branches.

I have also opened a PR upstream for an additional fasm fix which is still needed.

acomodi, 2019-06-19:
This can be closed as the repository has been reworked.

closed 2019-06-19 by acomodi