SymbiFlow/vtr-verilog-to-routing #72: Branch: avoid-criticality-issue

acomodi, 2019-06-07:
This branch is related to a timing issue that deals with an `infinite slack` issue.
Whenever the slack is infinite the loop continues and no Criticality error is produced. https://github.com/SymbiFlow/vtr-verilog-to-routing/blob/da4a6af1d33018617ed3c017bc67a47007272fa7/vpr/src/timing/timing_util.cpp#L574-L576

IMO this should not be merged upstream, as it is related to the fact that SymbiFlow is not yet including all necessary timing information.

This branch can be deleted as soon as timing information prevent `slack` to go to an infinite value.

acomodi, 2019-07-01:
@litghost I have tried not to include this patch in the local octopus merge and murax did compile without errors with the place_algorithm set to `path_timing_driven`. I am running `all_xc7` and `all_ice40` to see if this patch is still needed.

I guess that if both of the builds do complete successfully, we can discard this branch.

litghost, 2019-07-01:
Thanks for checking that, sounds good to me.  One less branch!