SymbiFlow/vtr-verilog-to-routing #75: Branch: place-skip-x-locations

acomodi, 2019-06-07:
This branch has changes that fix an issue during placement for which the placer cannot find any block in a `x` location. Without this, placer raises an assert.

#### Send upstream
This fix has to be sent upstream and there already is an open PR for it: https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/577

acomodi, 2019-06-07:
@litghost Can you add a space between the `if` and the `(` in your open PR (verilog-to-routing#577)?
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/cda09389f3efe27294d97b96c9bd759a838fdd5d/vpr/src/place/place.cpp#L2145

CI will fail because of format

litghost, 2019-06-07:
Rebased onto master and formatted, done.

acomodi, 2019-06-17:
@litghost looks like the PR CI has failed a QoR test:
```
regression_tests/vtr_reg_strong/strong_global_routing...[Fail] 
```

It should not be related to the changes, but it would be good to rebase on top so the CI is re-run. I think since it was run last time a few changes in the golden QoR have been applied.

acomodi, 2019-06-26:
This has been merged upstream and is part of conda packate, closing.

closed 2019-06-26 by acomodi