SymbiFlow/vtr-verilog-to-routing #76: Branch: placement-tolerance-errors-to-warnings

acomodi, 2019-06-07:
This branch demotes `errors` to `warnings`.

#### Do not send upstream
This changes should not be sent upstream. This because there is a WIP that allows to demote errors to warnings from a command line vpr option (https://github.com/SymbiFlow/vtr-verilog-to-routing/pull/66).
Once the mentioned PR is landed, this branch can be deleted.
Another way to delete this branch is also to solve the issue from a SymbiFlow point of view. IMO this will happen when correct timing information are added to the architecture.

---

##### Issue
There is an issue though with the error count:
https://github.com/SymbiFlow/vtr-verilog-to-routing/blob/840a5fd2496fed17719718cb3ea15b00eaca5029/vpr/src/place/place.cpp#L3546-L3556
The `error++` should not be increased if the error is treated as a warning.

acomodi, 2019-06-21:
This should actually go upstream as there is an issue with the counting of the errors.
I have modified the patch so that the error is demoted to warning (and `error++` is skipped) whenever --strict_checks option is disabled.

@litghost Do you think there is another way to treat this or is it ok to merge upstream?

litghost, 2019-06-25:
> @litghost Do you think there is another way to treat this or is it ok to merge upstream?

Unclear.  For now we need this otherwise VPR just stops.  Whether the underlying cause is numerical errors in VPR or deficiencies in the timing model is unclear.  I'm generally leaning towards numerical problems in VPR, and that this check does more harm than good.

acomodi, 2019-06-25:
@litghost I have discovered that we can overcome this error with this PR: https://github.com/SymbiFlow/symbiflow-arch-defs/pull/859, more info are there on the discoveries.

closed 2019-06-25 by litghost