SymbiFlow/vtr-verilog-to-routing #82: Branch: timing-graph-builder-fix

acomodi, 2019-06-07:
This branch includes a fix which solves a `tatum` error issue for some SymbiFlow tests. I am unsure on how this solves the problem as I have found the fix by diffing the old master+wip branch with the master.
The addition to VPR is the following:
https://github.com/SymbiFlow/vtr-verilog-to-routing/blob/89ebc9d121dd495ad958b11e155e6937b9c93a7a/vpr/src/timing/timing_graph_builder.cpp#L55

@litghost does this need to go upstream?

litghost, 2019-06-07:
This likely needs to go upstream, it might need to a behind a flag, or better detection.  I believe I changed this to support IO blocks (which had no combination sinks).

acomodi, 2019-06-07:
Ok, I will open a PR upstream and discuss it with vtr-devs

acomodi, 2019-06-10:
PR: https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/643

mgielda, 2019-06-16:
The PR was a bit preliminary since we do not fully understand why exactly this is needed - @litghost can you remember why you made this change in the first place (other than that it stops errors from occurring ;)) - the VTR devs seem to suggest it's an architectural issue we need to fix.

acomodi, 2019-06-19:
On the open PR it was suggested to use a command line option to workaround the issue (for the short term period). I have updated the PR. Will move this to "PR open with upstream" column.

acomodi, 2019-07-23:
In conda package and used in symbiflow-arch-defs. Closing

closed 2019-07-23 by acomodi