SymbiFlow/vtr-verilog-to-routing #108: Branch: revert-deterministic-packer-changes

acomodi, 2019-06-18:
### Why did we need this? (what does this change enable us to do)

This change is needed to make VtR correctly route again picosoc for ice40. The reason why picosoc could not route are not yet clear, but they are related to a different `packing` result.

### What did it change?

This patch reverts the following commits:
- https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/b716b85a6399503b592d62b9b877f030434546ff
- https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/aed1b24720f128eb558d92e2578c3e8a90e03def

Specifically, those commits were adding determinism to the packing stage, as it has been discovered by vtr-devs that different systems running VtR could produce different packing results.

### Should it be merged upstream - if not, when can we delete it?
No, this patch is a temporary fix that cannot be sent upstream and we will able to delete it as soon as it's clear why the packer output becomes unroutable for some of the SymbiFlow tests.

### What is needed to get this merged / deleted? 
Investigate the causes of the impossibility to route and solve them.

* [ ] is the implementation work to make suitable for merging / deletion completed?
* [ ] Is there an associated test? 
* [ ] is this currently part of the Conda package?
* [ ] is this properly cleaned up in our local repositories? 

### Tracker / branch / PR & other useful links
code diff: https://github.com/verilog-to-routing/vtr-verilog-to-routing/compare/master...SymbiFlow:revert-deterministic-packer-changes

acomodi, 2019-06-20:
@litghost turns out that reverting the changes in upstream that caused picosoc ice40 to fail some of the `strong_reg_tests` do fail making CI go red (see latest update of master+wip CI https://travis-ci.com/SymbiFlow/vtr-verilog-to-routing/jobs/209375381).
This is an unfortunate case as by not reverting those changes picosoc will fail, whereas reverting them will cause CI to fail. 
This is actually something we should look deeper into, as it could be an issue that slight changes to the packer can cause this kind of outcomes (both in SymbiFlow and both in VtR CI).

litghost, 2019-06-20:
> @litghost turns out that reverting the changes in upstream that caused picosoc ice40 to fail some of the `strong_reg_tests` do fail making CI go red (see latest update of master+wip CI https://travis-ci.com/SymbiFlow/vtr-verilog-to-routing/jobs/209375381).
> This is an unfortunate case as by not reverting those changes picosoc will fail, whereas reverting them will cause CI to fail.
> This is actually something we should look deeper into, as it could be an issue that slight changes to the packer can cause this kind of outcomes (both in SymbiFlow and both in VtR CI).
> 
> > **[Travis CI - Test and Deploy with Confidence](https://travis-ci.com/SymbiFlow/vtr-verilog-to-routing/jobs/209375381)**

Try what @elmsfu did here: https://github.com/SymbiFlow/symbiflow-arch-defs/pull/693#discussion_r295883606

Those arguments might resolving the issue without needing to revert the packer changes.

acomodi, 2019-06-20:
Makes sense actually, didn't think about that, thanks

acomodi, 2019-06-21:
The fix in https://github.com/SymbiFlow/symbiflow-arch-defs/pull/693#discussion_r295883606 makes this branch useless. Closing and deleting branch.

closed 2019-06-21 by acomodi