SymbiFlow/vtr-verilog-to-routing #136: Slow delta delays computation on master+wip

acomodi, 2019-06-28:
With the current master+wip (https://github.com/SymbiFlow/vtr-verilog-to-routing/commit/893663cd92dd9afe15557fa5a6f7a983a76f8626), the Titan benchmarks take more time than upstream master during `delta delay computations`.

Here are some timing data from two runs:
- Upstream master
```
# Placement
## Computing placement delta delay look-up
RR graph channel widths unchanged, skipping RR graph rebuild
### Computing delta delays
### Computing delta delays took 687.78 seconds (max_rss 2491.9 MiB, delta_rss +0.0 MiB)
## Computing placement delta delay look-up took 688.08 seconds (max_rss 2491.9 MiB, delta_rss +0.0 MiB)
```

- SymbiFlow master
```
# Placement
## Computing placement delta delay look-up
RR graph channel widths unchanged, skipping RR graph rebuild
### Computing delta delays
### Computing delta delays took 2446.69 seconds (max_rss 2531.6 MiB, delta_rss +0.0 MiB)
## Computing placement delta delay look-up took 2447.17 seconds (max_rss 2531.6 MiB, delta_rss +0.0 MiB)
```

acomodi, 2019-06-28:
@litghost FYI

litghost, 2019-06-28:
It is possible that:
https://github.com/SymbiFlow/vtr-verilog-to-routing/blob/master%2Bwip/vpr/src/route/router_lookahead_map.cpp#L134
and
https://github.com/SymbiFlow/vtr-verilog-to-routing/blob/master%2Bwip/vpr/src/route/router_lookahead_map_utils.cpp#L9

are in conflict in the linker.  The version isn map_utils and used by connection box utils uses the commented out logic from https://github.com/SymbiFlow/vtr-verilog-to-routing/blob/master%2Bwip/vpr/src/route/router_lookahead_map.cpp#L156

Putting all of https://github.com/SymbiFlow/vtr-verilog-to-routing/blob/master%2Bwip/vpr/src/route/router_lookahead_map_utils.h will demostrate on way or another if this is the issue.

acomodi, 2019-07-03:
Closing. Solved via https://github.com/SymbiFlow/vtr-verilog-to-routing/pull/140

closed 2019-07-03 by acomodi