SymbiFlow/vtr-verilog-to-routing #250: VTR cannot handle clustering of multiple nodes of the same type

kgugala, 2019-08-20:
V2X tests from https://github.com/SymbiFlow/symbiflow-arch-defs/pull/703 generate architectures from the test models and then tries to place and route the test models into the generated architecture. Since the architecture is generated from the model, the model should fit the device and use 100% of available resources.

The `multiple_instance` test uses the following verilog model https://github.com/SymbiFlow/symbiflow-arch-defs/blob/af78aa52b54d640071ea197c26c0021afe2df541/utils/vlog/tests/multiple_instance/multiple_instance.sim.v

The architecture is generated properly, but VPR cannot pack all the ADDERs into single cluster, resulting in exceeding the device resources.

The packing fail is caused by the way we check if the molecule to be packed is feasible to be packed to the cluster:

In https://github.com/SymbiFlow/vtr-verilog-to-routing/blob/master%2Bwip/vpr/src/pack/cluster_placement.cpp#L691
we only check if the atom we consider for packing is feasible to pack using available primitive in the cluster. This is done by simple check if the primitive and atom are the same type and if the primitive has enough pins to fit the atom.
In our case, since both possible cluster primitives are ADDERs, the first one is always chosen and removed from available list. This is, of course, a bug - we should not chose the first ADDER type for atoms of the second type.

Pin feasibility filter is able to catch the issue as soon as we run out of the first type of ADDERs.

This is a blocker for https://github.com/SymbiFlow/symbiflow-arch-defs/pull/703

kgugala, 2019-08-20:
@mithro @litghost PTAL

mithro, 2019-08-20:
@kgugala I think I'm missing something here? Are you saying that if you have multiple leafs pb_types which impliment the same blackbox it will never considers the second leaf as an option?

kgugala, 2019-08-20:
@mithro VPR considers the second is pb_type only after it uses all the first type ones.

The issue in the example is that the first type is used for packing molecules/atoms of both types.

mithro, 2019-08-20:
@kgugala Why is it a problem to only consider the second pb_type after all the first one is used?

kgugala, 2019-08-20:
@mithro pb_types in the example have different routing. The clusterer uses the first pb_type, as long as it is available, to pack both molecule types.

The problem is that using pb_type A for clustering atom type B is unfeasible (from routing perspective).

mithro, 2019-08-20:
@kgugala - Can a pack pattern solve things?

mithro, 2019-08-20:
@kgugala - The reason the routing is unfeasable is because of the internal routes between carry blocks?

kgugala, 2019-08-20:
@mithro pb_types in the example are part of a single pack pattern, unfortunately this does not help

mithro, 2019-08-20:
```
	/* n = 0..DATA_WIDTH
	 *
	 *             ↓      ↓      ↓
	 *   comb_apb(a[n] + b[n] + cin[n]) → sum[n]
	 *                                 ↓   
	 *                 ↓      ↓        ↓
	 *       comb_cpd(c[n] + d[n] + ......) → sum[4+n]
	 *                                     ↓
	 *                                   cout[n]
	 */
```

mithro, 2019-08-20:
I think my input verilog isn't really describing what I wanted...

mithro, 2019-08-20:
I think diagram at https://docs.google.com/drawings/d/1J4YTA-0MozQvMnBB-IhEQGBkgyVqIHEusXb1j-0G33M/edit explains what I was going for...

mithro, 2019-08-20:
[image]

mithro, 2019-08-20:
I can't remember which version I was trying to do.. I think the second one...

kgugala, 2019-08-25:
the verilog seems to be correct for the second diagram

kgugala, 2019-08-26:
The problem was caused by the way the architecture is generated. The first adder has it's COUT connected to both cluster output and input of the next adder. VPR was always choosing the first ADDER atom because it was in fact feasible for both possible types.

This commit fixes that https://github.com/SymbiFlow/symbiflow-arch-defs/pull/703/commits/e0e0541dab313323639dd76c58fab2a2c48b6f55

closed 2019-08-26 by kgugala