SymbiFlow/vtr-verilog-to-routing #263: Branch: optimize-ndmatrix

acomodi, 2019-09-10:
### Why did we need this? (what does this change enable us to do)

Previous implementation recomputed dimension strides every usage.  When using the map lookahead, this has a very significant affect, as the core of the map lookahead is an NdMatrix.

### What did it change?

### Should it be merged upstream - if not, when can we delete it?

### What is needed to get this merged / deleted? 

* [ ] is the implementation work to make suitable for merging / deletion completed?
* [ ] Is there an associated test? 
* [ ] is this currently part of the Conda package?
* [ ] is this properly cleaned up in our local repositories? 

### Tracker / branch / PR & other useful links

acomodi, 2019-09-10:
@litghost can you please provide more info on what this patch changes and if it is or will be sent upstream?

litghost, 2019-10-02:
Merged upstream.

closed 2019-10-02 by litghost