SymbiFlow/vtr-verilog-to-routing #281: Connection box lookahead segment search is fixed

litghost, 2019-09-26:
#### Expected Behaviour

When the connection box and map lookahead sample segments, they are robust in their search for segments and gaining diversity of the grid delays.

#### Current Behaviour

Both the connection box and map lookaheads use fixed grid locations:
 - Map lookahead : https://github.com/SymbiFlow/vtr-verilog-to-routing/blob/master%2Bwip/vpr/src/route/router_lookahead_map.cpp#L33
 - https://github.com/SymbiFlow/vtr-verilog-to-routing/blob/master%2Bwip/vpr/src/route/connection_box_lookahead_map.cpp#L332

#### Possible Solution

Both lookaheads should have an ability to search the grid for segments and some form of configurable "sampling effort".  For example, the current connection box sampling is likely excessive and causes the lookahead runtime to be excessive.  A smarter sampling strategy will likely yield similar lookahead quality and reduced runtime.

#### Steps to Reproduce

1. Shift CMT from upper right to lower left
2. During lookahead generation, many segments fail to get sampled (http://sandbox.hackerfoo.com:8008/_matrix/media/v1/download/sandbox.hackerfoo.com/fEqtJbKgwPKyhbOUJwVqfNnz)

litghost, 2019-09-26:
Note, that if a generalized sampling algorithm is problematic, or fragile, then one potential answer is simply to move the sampling locations to a command line argument.  This potentially could be "for these segments, sample from these locations", allowing variable diversity for the segments.

HackerFoo, 2019-12-06:
https://github.com/SymbiFlow/vtr-verilog-to-routing/pull/343

closed 2019-12-06 by HackerFoo