SymbiFlow/vtr-verilog-to-routing #295: Equivalent tiles implementation assumes that a pb_type has a physical with the same name

litghost, 2019-10-11:
#### Expected Behaviour

pb_type names and physical types names do not have to be the same.

#### Current Behaviour

VTR segfauls at `vpr/src/base/read_netlist.cpp:956` if the logical pb_type name does not have a physical tile with the same name.

#### Possible Solution

Netlist should have not have any physical tiles.

#### Steps to Reproduce

1. Build target `file_xc7_archs_artix7_devices_rr_graph_xc7a50t_test.rr_graph.virt.xml` from https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1072/commits/3b4c5f8c5d1b458dc13b695edd5933d0fb0b2d03 , segfaults at  `vpr/src/base/read_netlist.cpp:956`

#### Context

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

litghost, 2019-10-11:
There are other instances of nullptr deferences in the equivalent tile code, please go back and check for nullptr whenever a new pointer is reached.

Second example of nullptr deference I found:
https://github.com/SymbiFlow/vtr-verilog-to-routing/blob/wip/equivalent-sites/vpr/src/base/SetupGrid.cpp#L670

litghost, 2019-10-11:
FYI @acomodi

acomodi, 2019-10-14:
@litghost I have updated `wip/equivalent-sites` to fix this issue. It still needs some work, but it shouldn't block anymore the work on https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1072.

acomodi, 2019-10-14:
Using https://github.com/SymbiFlow/symbiflow-arch-defs/commit/3b4c5f8c5d1b458dc13b695edd5933d0fb0b2d03 I noticed that there was a pb_type (`BLK_TL-IOPAD_S`) that did not have a corresponding tile. Therefore, I have also added a check during the XML parsing to ensure the presence of at least one physical tile for every `pb_type`

litghost, 2019-10-14:
> @litghost I have updated `wip/equivalent-sites` to fix this issue. It still needs some work, but it shouldn't block anymore the work on [SymbiFlow/symbiflow-arch-defs#1072](https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1072).

I see that.  If I'm able to successfully place and route IO's through multiple IO types (e.g. R/L, M/S), then I'll integrate a new master+wip with the lastest `wip/equivalent-sites`.

acomodi, 2019-10-23:
Solved in `wip/equivalent-sites`:  https://github.com/SymbiFlow/vtr-verilog-to-routing/commit/bdc84627339b662eb2ffcd12f2c2caaaffce0b0b

closed 2019-10-23 by acomodi