SymbiFlow/vtr-verilog-to-routing #302: Equivalent tiles needs to be run with Sanitizers

acomodi, 2019-10-17:
The equivalent tiles branch could have some memory-related issues.
Equivalent tiles branch should be run with the sanitizers switched on to have a complete overview on what and where could the issues be.

acomodi, 2019-10-17:
@litghost FYI

litghost, 2019-10-17:
Setting VTR_ENABLE_SANITIZE to ON in the CMakeCache.txt will enable the asan and ubsan for GCC.  Let's start with running tests with those enabled, including some Symbiflow tests.

Once that is done, we can talk about how to get sanitizers running on Clang.

litghost, 2019-10-17:
Clang bug: https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/1009

litghost, 2019-10-17:
Additional tests should be added to the VTR test suite that exercise equivilant tiles like how we are using it in Symbiflow, example:
 - Having a logical block type that has no physical block with the same name
 - Having direct connections and placer macros exist between physical blocks that have no logical block type with the same name
 - Having many more # of physical blocks than logical blocks

acomodi, 2019-10-18:
Ok, I will start to prepare test architectures to check all these possibilities and add them to the `vtr_strong_reg` suite.

I am also running all the tests with the Sanitizers switched on

acomodi, 2019-10-18:
By running VtR reg tests this leak showed up in most of them:

```
Direct leak of 60312 byte(s) in 4 object(s) allocated from:
    #0 0x7f5ac5b19618 in operator new[](unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xe0618)
    #1 0x56301771b835 in alloc_and_load_pb_graph_pin_lookup_from_index(t_logical_block_type const*) /home/alessandro/projects/symbiflow/tmp/vtr-comodi/vpr/src/util/vpr_utils.cpp:1342
    #2 0x563017705e8a in IntraLbPbPinLookup::IntraLbPbPinLookup(std::vector<t_logical_block_type, std::allocator<t_logical_block_type> > const&) /home/alessandro/projects/symbiflow/tmp/vtr-comodi/vpr/src/util/vpr_utils.cpp:277
    #3 0x563016f573d3 in ClbDelayCalc::ClbDelayCalc() /home/alessandro/projects/symbiflow/tmp/vtr-comodi/vpr/src/timing/clb_delay_calc.inl:10
    #4 0x563016f5814c in PostClusterDelayCalculator::PostClusterDelayCalculator(AtomNetlist const&, AtomLookup const&, vtr::vector<vtr::StrongId<cluster_net_id_tag, int, -1>, float*>&) /home/alessandro/projects/symbiflow/tmp/vtr-comodi/vpr/src/timing/PostClusterDelayCalculator.tpp:27
    #5 0x56301701c8dc in void __gnu_cxx::new_allocator<PostClusterDelayCalculator>::construct<PostClusterDelayCalculator, AtomNetlist const&, AtomLookup const&, vtr::vector<vtr::StrongId<cluster_net_id_tag, int, -1>, float*>&>(PostClusterDelayCalculator*, AtomNetlist const&, AtomLookup const&, vtr::vector<vtr::StrongId<cluster_net_id_tag, int, -1>, float*>&) /usr/include/c++/7/ext/new_allocator.h:136
    #6 0x563017019b9a in void std::allocator_traits<std::allocator<PostClusterDelayCalculator> >::construct<PostClusterDelayCalculator, AtomNetlist const&, AtomLookup const&, vtr::vector<vtr::StrongId<cluster_net_id_tag, int, -1>, float*>&>(std::allocator<PostClusterDelayCalculator>&, PostClusterDelayCalculator*, AtomNetlist const&, AtomLookup const&, vtr::vector<vtr::StrongId<cluster_net_id_tag, int, -1>, float*>&) /usr/include/c++/7/bits/alloc_traits.h:475
    #7 0x563017013e88 in std::_Sp_counted_ptr_inplace<PostClusterDelayCalculator, std::allocator<PostClusterDelayCalculator>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<AtomNetlist const&, AtomLookup const&, vtr::vector<vtr::StrongId<cluster_net_id_tag, int, -1>, float*>&>(std::allocator<PostClusterDelayCalculator>, AtomNetlist const&, AtomLookup const&, vtr::vector<vtr::StrongId<cluster_net_id_tag, int, -1>, float*>&) /usr/include/c++/7/bits/shared_ptr_base.h:526
    #8 0x563017008dd9 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<PostClusterDelayCalculator, std::allocator<PostClusterDelayCalculator>, AtomNetlist const&, AtomLookup const&, vtr::vector<vtr::StrongId<cluster_net_id_tag, int, -1>, float*>&>(std::_Sp_make_shared_tag, PostClusterDelayCalculator*, std::allocator<PostClusterDelayCalculator> const&, AtomNetlist const&, AtomLookup const&, vtr::vector<vtr::StrongId<cluster_net_id_tag, int, -1>, float*>&) /usr/include/c++/7/bits/shared_ptr_base.h:637
    #9 0x563016ff5867 in std::__shared_ptr<PostClusterDelayCalculator, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<PostClusterDelayCalculator>, AtomNetlist const&, AtomLookup const&, vtr::vector<vtr::StrongId<cluster_net_id_tag, int, -1>, float*>&>(std::_Sp_make_shared_tag, std::allocator<PostClusterDelayCalculator> const&, AtomNetlist const&, AtomLookup const&, vtr::vector<vtr::StrongId<cluster_net_id_tag, int, -1>, float*>&) /usr/include/c++/7/bits/shared_ptr_base.h:1295
    #10 0x563016fdcac2 in std::shared_ptr<PostClusterDelayCalculator>::shared_ptr<std::allocator<PostClusterDelayCalculator>, AtomNetlist const&, AtomLookup const&, vtr::vector<vtr::StrongId<cluster_net_id_tag, int, -1>, float*>&>(std::_Sp_make_shared_tag, std::allocator<PostClusterDelayCalculator> const&, AtomNetlist const&, AtomLookup const&, vtr::vector<vtr::StrongId<cluster_net_id_tag, int, -1>, float*>&) /usr/include/c++/7/bits/shared_ptr.h:344
    #11 0x563016fbfe06 in std::shared_ptr<PostClusterDelayCalculator> std::allocate_shared<PostClusterDelayCalculator, std::allocator<PostClusterDelayCalculator>, AtomNetlist const&, AtomLookup const&, vtr::vector<vtr::StrongId<cluster_net_id_tag, int, -1>, float*>&>(std::allocator<PostClusterDelayCalculator> const&, AtomNetlist const&, AtomLookup const&, vtr::vector<vtr::StrongId<cluster_net_id_tag, int, -1>, float*>&) /usr/include/c++/7/bits/shared_ptr.h:691
    #12 0x563016f92248 in std::shared_ptr<PostClusterDelayCalculator> std::make_shared<PostClusterDelayCalculator, AtomNetlist const&, AtomLookup const&, vtr::vector<vtr::StrongId<cluster_net_id_tag, int, -1>, float*>&>(AtomNetlist const&, AtomLookup const&, vtr::vector<vtr::StrongId<cluster_net_id_tag, int, -1>, float*>&) /usr/include/c++/7/bits/shared_ptr.h:707
    #13 0x563016f2f118 in vpr_route_flow(t_vpr_setup&, t_arch const&) /home/alessandro/projects/symbiflow/tmp/vtr-comodi/vpr/src/base/vpr_api.cpp:658
    #14 0x563016f27e6e in vpr_flow(t_vpr_setup&, t_arch&) /home/alessandro/projects/symbiflow/tmp/vtr-comodi/vpr/src/base/vpr_api.cpp:366
    #15 0x563016e1ba7b in main /home/alessandro/projects/symbiflow/tmp/vtr-comodi/vpr/src/main.cpp:61
    #16 0x7f5ac2219b96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)

```

I need to check whether this is introduced with the `wip/equivalent_sites` PR or it was already present

acomodi, 2019-10-18:
I have checked, and VtR master at https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/2df20a2e7c1d22850378ddf1f06e2e1267f6bcf9 results in the same Leakage errors.

I will now run it against some of the SymbiFlow tests as well.

litghost, 2019-10-18:
I found a segfault yesterday, and turning on ASAN found this bug:
```
=================================================================                                                                                                                                                 
==109912==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6040045a29c0 at pc 0x0000014c1eba bp 0x7ffdfe457930 sp 0x7ffdfe457928                                                                        
READ of size 4 at 0x6040045a29c0 thread T0                                                                                                                                                                        
    #0 0x14c1eb9 in print_stats() /home/keithrothman/cat_x/vtr-verilog-to-routing/vpr/src/pack/output_clustering.cpp:93:62                                                                       
    #1 0x14b687d in output_clustering(vtr::vector<vtr::StrongId<cluster_block_id_tag, int, -1>, std::vector<t_intra_lb_net, std::allocator<t_intra_lb_net> >*> const&, bool, std::unordered_set<vtr::StrongId<atom_
net_id_tag, int, -1>, std::hash<vtr::StrongId<atom_net_id_tag, int, -1> >, std::equal_to<vtr::StrongId<atom_net_id_tag, int, -1> >, std::allocator<vtr::StrongId<atom_net_id_tag, int, -1> > > const&, std::__cxx11
::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, bool) /home/keithrothman/cat_x/vtr-verilog-to-routing/vpr/src/pack/output_clustering.cpp:612:5          
    #2 0x1365ee1 in do_clustering(t_packer_opts const&, t_analysis_opts const&, t_arch const*, t_pack_molecule*, int, std::unordered_set<vtr::StrongId<atom_net_id_tag, int, -1>, std::hash<vtr::StrongId<atom_net_
id_tag, int, -1> >, std::equal_to<vtr::StrongId<atom_net_id_tag, int, -1> >, std::allocator<vtr::StrongId<atom_net_id_tag, int, -1> > > const&, std::multimap<vtr::StrongId<atom_block_id_tag, int, -1>, t_pack_mol
ecule*, std::less<vtr::StrongId<atom_block_id_tag, int, -1> >, std::allocator<std::pair<vtr::StrongId<atom_block_id_tag, int, -1> const, t_pack_molecule*> > >&, std::unordered_map<vtr::StrongId<atom_block_id_tag
, int, -1>, t_pb_graph_node*, std::hash<vtr::StrongId<atom_block_id_tag, int, -1> >, std::equal_to<vtr::StrongId<atom_block_id_tag, int, -1> >, std::allocator<std::pair<vtr::StrongId<atom_block_id_tag, int, -1>
const, t_pb_graph_node*> > > const&, bool, bool, std::vector<t_lb_type_rr_node, std::allocator<t_lb_type_rr_node> >*, t_ext_pin_util_targets const&, t_pack_high_fanout_thresholds const&) /home/k
eithrothman/cat_x/vtr-verilog-to-routing/vpr/src/pack/cluster.cpp:771:5                                                                                                                                           
    #3 0x8a4b36 in try_pack(t_packer_opts*, t_analysis_opts const*, t_arch const*, t_model const*, t_model const*, float, std::vector<t_lb_type_rr_node, std::allocator<t_lb_type_rr_node> >*) /ho
me/keithrothman/cat_x/vtr-verilog-to-routing/vpr/src/pack/pack.cpp:125:35                                                                                                                                         
    #4 0x667a88 in vpr_pack(t_vpr_setup&, t_arch const&) /home/keithrothman/cat_x/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp:549:12                                                         
    #5 0x65e32e in vpr_pack_flow(t_vpr_setup&, t_arch const&) /home/keithrothman/cat_x/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp:475:22                                                    
    #6 0x65dcaa in vpr_flow(t_vpr_setup&, t_arch&) /home/keithrothman/cat_x/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp:346:29                                                               
    #7 0x51317d in main /home/keithrothman/cat_x/vtr-verilog-to-routing/vpr/src/main.cpp:61:31                                                                                                   
    #8 0x7f7f93d4f52a in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2352a)
    #9 0x438469 in _start (/home/keithrothman/cat_x/vtr-verilog-to-routing/build/vpr/vpr+0x438469)                                                                                               

0x6040045a29c0 is located 8 bytes to the right of 40-byte region [0x6040045a2990,0x6040045a29b8)
allocated by thread T0 here:
    #0 0x4e04aa in calloc (/home/keithrothman/cat_x/vtr-verilog-to-routing/build/vpr/vpr+0x4e04aa)                                                                                               
    #1 0x19a6d26 in vtr::calloc(unsigned long, unsigned long) /home/keithrothman/cat_x/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_memory.cpp:42:16                                           
    #2 0x14bfeda in print_stats() /home/keithrothman/cat_x/vtr-verilog-to-routing/vpr/src/pack/output_clustering.cpp:57:34                                                                       
    #3 0x14b687d in output_clustering(vtr::vector<vtr::StrongId<cluster_block_id_tag, int, -1>, std::vector<t_intra_lb_net, std::allocator<t_intra_lb_net> >*> const&, bool, std::unordered_set<vtr::StrongId<atom_
net_id_tag, int, -1>, std::hash<vtr::StrongId<atom_net_id_tag, int, -1> >, std::equal_to<vtr::StrongId<atom_net_id_tag, int, -1> >, std::allocator<vtr::StrongId<atom_net_id_tag, int, -1> > > const&, std::__cxx11
[... cut ...]
    #9 0x51317d in main /home/keithrothman/cat_x/vtr-verilog-to-routing/vpr/src/main.cpp:61:31                                                                                                   
    #10 0x7f7f93d4f52a in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2352a)                                                        ```

`output_clustering.cpp` appears to be tracking arrays as logical tiles (which I think is correct) but using indicies from physical tiles.  Why would a packing stage function need to using a physical tile at all?

acomodi, 2019-10-21:
I get the following compilation error when building `master+wip` vpr with sanitizers enabled:
```
[  0%] Built target libpugixml
[  1%] Built target libcatch
[  1%] Built target libargparse
[  3%] Built target libsdcparse
[  3%] Linking CXX executable sdcparse_test
In function ‘sdc_parse_file’:
lto1: fatal error: Reference statement index not found
compilation terminated.
lto-wrapper: fatal error: /usr/bin/c++ returned 1 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
libs/EXTERNAL/libsdcparse/CMakeFiles/sdcparse_test.dir/build.make:84: recipe for target 'libs/EXTERNAL/libsdcparse/sdcparse_test' failed
make[3]: *** [libs/EXTERNAL/libsdcparse/sdcparse_test] Error 1
CMakeFiles/Makefile2:525: recipe for target 'libs/EXTERNAL/libsdcparse/CMakeFiles/sdcparse_test.dir/all' failed
make[2]: *** [libs/EXTERNAL/libsdcparse/CMakeFiles/sdcparse_test.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:71: recipe for target 'all' failed
make: *** [all] Error 2

```

litghost, 2020-03-04:
Equivalent tiles has been merged upstream.

closed 2020-03-04 by litghost