SymbiFlow/vtr-verilog-to-routing #305: Build breaks with the latest abc: error: cast from pointer to smaller type 'ABC_PTRINT_T' (aka 'int') loses information

yurivict, 2019-10-26:
```
In file included from /usr/local/include/abc/misc/vec/vec.h:29:
/usr/local/include/abc/misc/util/abc_global.h:314:77: error: cast from pointer to smaller type 'ABC_PTRINT_T' (aka 'int') loses information
static inline int      Abc_Ptr2Int( void * p )                { return (int)(ABC_PTRINT_T)p;      }
                                                                            ^~~~~~~~~~~~~~~
/usr/local/include/abc/misc/util/abc_global.h:316:78: error: cast from pointer to smaller type 'ABC_PTRUINT_T' (aka 'unsigned int') loses information
static inline word     Abc_Ptr2Wrd( void * p )                { return (word)(ABC_PTRUINT_T)p;    }
                                                                             ^~~~~~~~~~~~~~~~
/usr/local/include/abc/misc/util/abc_global.h:333:43: warning: '__APPLE__' is not defined, evaluates to 0 [-Wundef]
#if (defined(LIN) || defined(LIN64)) && !(__APPLE__ & __MACH__) && !defined(__MINGW32__)
                                          ^
/usr/local/include/abc/misc/util/abc_global.h:333:55: warning: '__MACH__' is not defined, evaluates to 0 [-Wundef]
#if (defined(LIN) || defined(LIN64)) && !(__APPLE__ & __MACH__) && !defined(__MINGW32__)
                                                      ^
```

clang-8 on FreeBSD.

litghost, 2019-10-28:
This issue and the other build failures appear to be present in upstream VTR (https://github.com/verilog-to-routing/vtr-verilog-to-routing), so this issue should be tracked and fixed there.   Symbiflow/vtr-verilog-to-routing is a downstream fork, and changes like this should be made upstream, and Symbiflow will integrate them.

yurivict, 2019-10-28:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/1017

closed 2019-10-28 by yurivict