SymbiFlow/vtr-verilog-to-routing #335: Branch: place-constraints

acomodi, 2019-11-28:
### Why did we need this? (what does this change enable us to do)

We need this to allow VPR using and detecting user-defined placement constraints to allow also the fixed placement of blocks different from IOs.

### What did it change?

Initial placement

### Should it be merged upstream - if not, when can we delete it?
No. This is a temporary workaround to solve https://github.com/SymbiFlow/symbiflow-arch-defs/issues/1182. When https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/1045 gets solved, we won't need this patch anymore

### What is needed to get this merged / deleted? 

* [ ] is the implementation work to make suitable for merging / deletion completed?
* [ ] Is there an associated test? 
* [ ] is this currently part of the Conda package?
* [ ] is this properly cleaned up in our local repositories? 

### Tracker / branch / PR & other useful links

litghost, 2020-03-04:
I've opened an upstream PR, as this branch is generally useful: https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1189

acomodi, 2020-03-11:
This has been merged upstream. closing

closed 2020-03-11 by acomodi