SymbiFlow/vtr-verilog-to-routing #524: Net alias not properly considered when creating set_{min,max}_delay constrainst

smunaut, 2020-06-24:
Alias are not considered when processing the `get_clocks` call when doing min/max delay constrainsts in the SDC.

#### Expected Behaviour

In the input netlist where you have theses aliases :

```
.names usb_I.uc_clk clk_wb
1 1
.names usb_I.tx_pkt_I.crc_16_I.clk clk_usb
1 1
```

Then the constraints : 

```
set_max_delay 40 -from [get_clocks {clk_wb}] -to [get_clocks {clk_usb}]
```

and

```
set_max_delay 40 -from [get_clocks {usb_I.tx_pkt_I.crc_16_I.clk}] -to [get_clocks {usb_I.uc_clk}]
```

should be equivalent. But in reality, only the latter works because aliases are not considered.

#### Current Behaviour

The SDC parsing fails with `set_max_path must specify at least one -from or -to clock` because it doesn't find any clocks.

#### Possible Solution

Fix https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/timing/read_sdc.cpp#L896 to also process net aliases like it was done for `create_clock` by @acomodi .

#### Steps to Reproduce

Try to setup any max/min delay constrainst using the non-primary clock name from the EBLIF.

#### Context

I'd like to setup constrainst with a stable clock net name and not some random name that was randomly picked among all the possible aliases by the synthesis tool.

#### Your Environment
* VTR revision used: 8980e46218542888fac879961b13aa7b0fba8432
* Operating System and version: Linux
* Compiler version: gcc 9.2.0

mithro, 2020-06-24:
We should send this to the upstream VtR repo at https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues -- @smunaut if you don't get to it I'm sure @acomodi will send it up.

acomodi, 2020-06-25:
Opened the issue upstream: https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/1379.
I added a fix locally, I need to add a basic regression test and I'll open a PR soon.

acomodi, 2020-06-25:
@smunaut I opened a PR with a fix that should solve the problem opened here: https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1380

smunaut, 2020-06-26:
@acomodi I just cherry picked that commit on top of the `master+wip` branch of Symbiflow and it works, thanks !

closed 2024-05-16 by smunaut