SymbiFlow/vtr-verilog-to-routing #537: Analysis option in VTR does not generate Blackbox timing in SDF correctly

tpagarani, 2020-07-08:
There are multiple issues related to SDF writer for Blackbox primitive

1. The primitives being modeled as Blackbox have combinational and sequential timings. In present --analysis code, blackbox combinational and sequential timings are not getting populated

2. Blackbox could have more than 1 reference clocks (e.g. Dual Port RAM). Current SDF generation does not handle timing checks w.r.t. specific clock. It uses a generic name "clk" to write out timing checks

Solution would be to traverse through all output and inputs pins of blackbox primitive to collect timing paths and 
timing checks. Also store the correct reference clock during the traversal.

mithro, 2020-07-08:
Should the timing information be provided by the black box implementation rather than vpr?

mithro, 2020-07-08:
@kgugala / @acomodi

tpagarani, 2020-07-08:
@mithro, well the reason I mention VPR is that sdf writer is part of VPR. Timing graph already has timing arcs and setup/hold, it's just that it's not being written out in SDF. I have already coded a fix in netlist_writer and in the process of creating a PR.

mithro, 2020-07-08:
@tpagarani - Not opposed to seeing this happening, just wondering if that is a work around or solution which results in less duplication in the output?